srg74/WLED-wemos-shield #116: Multiple antivirus scanners have flagged the files `wwindows.zip` and `ESP-Flasher.exe` as potentially malicious. This raises concerns about the safety and integrity of these files.

brammittendorff, 2024-01-04:
### Affected Files
1. `wwindows.zip`
   - VirusTotal link: [wwindows.zip analysis](https://www.virustotal.com/gui/file/406af8122dafa645503730bf2737e0bff72a00721d83657122b96981108af34d)
2. `ESP-Flasher.exe`
   - VirusTotal link: [ESP-Flasher.exe analysis](https://www.virustotal.com/gui/file/4cb6874972cd54a8644d2e5156be408230232b4733f54407d56922ab4d31a381)

### Malware Scan Results
- `wwindows.zip`: Detected as malicious by 5 out of 73 antivirus scanners.
- `ESP-Flasher.exe`: Detected as malicious by 5 security vendors. Notably, it was flagged by Bkav Pro, Cynet, DeepInstinct, SecureAge, and Skyhigh (SWG).

### Concerns
Given these detections, there is a concern that these files may contain harmful components that could compromise user security. Users of this repository should be made aware of these potential risks.

### Request
We kindly request an investigation into these files to verify their safety. If they are indeed compromised, we suggest removing or replacing them to ensure the security of all users.

### Additional Information
- Last Analysis Date: [Insert Date]
- File Sizes:
  - `wwindows.zip`:  15.2 MB 
  - `ESP-Flasher.exe`: 15.46 MB

Thank you for your attention to this matter. Ensuring the security and integrity of software is paramount, and your assistance is greatly appreciated.

srg74, 2024-01-04:
This software was used by multiple users and never has been reported as a virus.

closed 2024-01-27 by github-actions[bot]

brammittendorff, 2024-01-31:
> This software was used by multiple users and never has been reported as a virus. Most likely it's a false alarm.

Alright, just a heads up then.