srg74/WLED-wemos-shield #117: Question: PCBWay order with assembly

stanislavhannes, 2024-02-26:
Hey, first of all. Thank u very much for your work! :)
I was just wondering if it is possible to use the assembly service from PCBWay?
Obviously you have a SMD version of the board, but will it work with the service from the seller?
Thanks in advance

srg74, 2024-02-27:
PCBway have assembly service and you can use BOM file for reference. Obviously BOM file must be adjusted to parts they stock.

stanislavhannes, 2024-02-27:
Ok I understand. Anything else I have to be aware of? The PCB Layout e.g.? There are 6 different layouts. What is the difference between them?

srg74, 2024-02-27:
On [PCBway](https://www.pcbway.com/project/shareproject/WLED_wemos_shield.html) I have shared project with latest PCB design. All parts you need to use in your project is depends on your configuration. About configuration you can read here https://github.com/srg74/WLED-wemos-shield/wiki. Another option is to buy fully assembled shield from [Tindie](https://www.tindie.com/products/22307/) For further discussion better to join WLED Discord channel. My nick there is @Serg74

stanislavhannes, 2024-02-28:
Ok, so on PCBWay is online the one version without SMD parts online? The SMD version I can online buy on Tindie, right?

stanislavhannes, 2024-02-28:
Ok got it. Thanks

closed 2024-02-28 by stanislavhannes