srg74/WLED-wemos-shield #120: Plugin LEDs creates a shortcut on v3

gjornod, 2024-07-16:
After assembling and performing the appropriate checks, I power (5V) the shield successfully.
I then configure the appropriate pin to control the relay.
I can turn off and on my ESP8266 module with the push button and hear the relay working properly.
The issue starts when I add an LED strip. 
If the relay is on, I can light the strip. However, if I turn off and and then on (with the push button), it immediately creates a shortcut which restarts my power supply.
The system then starts to turn on, open the relay, and then shut down because of the short circuit infinitely, until I put a jumper between the power supply of the shield and the power supply of the strip (direclty on the molex or from the solder trace).

Things I tried to figure out what was wrong:

- I have soldered two different PCBs
- I have tried on ESP8266 wemos D1 mini and one ESP32 minikit wemos (I changed the relay and button pins in the config before starting)
- I tried to change the relay multiple times, I have TE as recommended but also Tongling.
- I have used different (out of the box) strips, all WS2812B though.

Did I miss something in the configuration?
I am running out of things to try and would greatly appreciate any guidance.

srg74, 2024-07-17:
Looks like you have power supply overloaded. If you using relays listed in BOM everything should be fine.

gjornod, 2024-07-17:
I used an industrial power supply and indeed the problem is solved. Thanks for the quick answer, I indeed  forgot to try another power supply.
I have been using the power supply from the previous setup for 100+ LEDs and the same relay. Do you have any idea why for 30 LEDs here I need a more powerful power supply?
Anyway, thank you for your great contribution, this is a very nice project.

srg74, 2024-07-17:
All depends on type of LED. Very important to use reliable power supply from reputable manufacturers. Cheap power supplies very often don't produce sufficient amounts of power. The best if you visit [WLED discord](https://discord.gg/QAh7wJHrRM) channel and read posts from users who already experience similar problems. Plus you could find very nice projects there too.

closed 2024-08-04 by gjornod