srg74/WLED-wemos-shield #123: Relay dows not switch off

spad, 2024-12-05:
My setup:
- universal shield with relay
- Wemos ESP32
- WLED 1.4.4

When I power the shield, the relay is off. Pushing the button, the relay switch to on but never switch off when the button is pressed again , nor when switching the "Power" menu in the UI. The LEDs, btw, are working as expected.

Using the board without shield, IO019 has level 0 and 3.2V when pressing the Power in the UI..
Wen mounted on the shield these levels are 0V and 2.5V

srg74, 2024-12-05:
Check in the relay settings first

spad, 2024-12-05:
I'm using v3.0 from pcbway.
I already cheked the build and the components values. All the components are from DigiKey from the BOM.
[image]

srg74, 2024-12-08:
Your board looks ok. Signal levels for relay are ok. Only things could be wrong is MosFET and diode.

closed 2024-12-26 by github-actions[bot]