srg74/WLED-wemos-shield #124: PCB Files Available?

jace055, 2024-12-09:
Hi There!

Are the files for the PCB available?  I would like to modify your (excellent) design to remove the relay and add a DC/DC converter to run the board and the lights off of a 24v power supply.  I am currently using [these ](https://www.amazon.com/dp/B01MQGMOKI?_encoding=UTF8&psc=1&ref_=cm_sw_r_cp_ud_dp_VW6ZRCTA8ZP20Y62DWXB) converters.  They are only 3A but I am only using them to run the lights on my 3d printers with your board.  My current draw is 700ma (Prusa) and 1.5a (Voron).  I would also like to add an additional mini fuse holder for the incoming power to the converter.  

I've built 15 of your boards so far for use around my home!  Everything from my Terminator arm lamp (modified base to fit your board on Printables) to my Christmas lights is powered by your board.  Much thanks!

srg74, 2024-12-09:
Board actually has DC/DC converter up to 24V. Schematic is available in resources.

jace055, 2024-12-09:
I was under the impression the output of the DC/DC converter was only sent to the D1 board.  The input power and voltage would be sent to the LED strip.  Was I wrong?  I thought this option was for powering the D1 when using 12v light strips.

srg74, 2024-12-09:
Voltage supplying board the same as voltage for LED strip. On board dc/dc converter is for esp circuit.

jace055, 2024-12-09:
That's what I thought.  Thanks.  

I didn't realize PCBWay paid out commissions.  Had I known that I wouldn't have asked for your PCB layout.   

Thanks again!  Great board!

closed 2024-12-09 by jace055