srg74/WLED-wemos-shield #128: Relay not switching

NXTler, 2025-04-27:
Greetings,
yesterday I assembled my first controller and began testing. Everything seems to work nicely so far, and I'm very impressed with overall quality. However, for reasons unknown to me, the Relay does not seem to switch at all and is permanently off. Everything is as the BOM suggests, except for SRD-05VDCSL-C (swapped with J1071CS5VDC.36) and 1N4148-P-TAP (swapped with 1N4150TR as per issue #97 ) with should be identical.
I checked all solder connections and component orientation so far. The only odd thing I noticed was that the mosfet's gate always had about 2.8 V (respective to ground) applied regardless of wled settings.

srg74, 2025-04-27:
Did you check pin settings for a relay?

NXTler, 2025-04-27:
Yes, the correct GPIO pin is selected, I have also checked if it's correct for my ESP and that I have continuity.

NXTler, 2025-05-13:
Are there any specific measurements that I could take to ensure that the mosfet is functional, for example?

srg74, 2025-05-13:
There is no special way to do it. Simple tutorial https://www.youtube.com/watch?v=gloikp9t2dA
Circuit I used is very common and everything should work. There is something you have overlooked.

NXTler, 2025-05-14:
Okay, so I took some measurements.
The LEDs are turned off: 0V are applied to the Gate and the difference between the Source and the Drain is 5,05V.
The LEDs are tuned on: 2,75V are applied to the Gate and the difference between the Source and the Drain is about 3,75.
When the LEDs are on, and I briefly apply 5V to the Mosfets gate, the Relay switches and stays in position until I turn off the LEDs.

What I gather from this is, that I need to somehow increase the voltage applied to the Mosfet's gate to have proper Relay switching.

closed 2025-06-02 by github-actions[bot]