srg74/WLED-wemos-shield #129: Вітаю

skipper4208, 2025-08-24:
Вітаю! есть пару тупый вопросов). discord skipper_2408. 2408skipper@gmail.com

skipper4208, 2025-08-24:
> So where is questions?

У меня есть esp32 wroom 32D на devkitv1 прошитая wled_DMX, max485 и лента
rgbw с чипом UCS512C4. Не могу найти адекватной инструкции подключения max
к esp и настройки DMX-out. Увидел аватарку и имя, понял что земляк

srg74, 2025-08-24:
All info is [here](https://kno.wled.ge/interfaces/dmx-output/)

skipper4208, 2025-08-24:
я читала!!!!

skipper4208, 2025-08-24:
надеюсь мы с тобой в акопах встретимся! Там и поговорим!

srg74, 2025-08-24:
Так що не зрозуміло?

skipper4208, 2025-08-24:
не зрозуміло чому не працюєю. GPIO2 - DI. max 5v земли все вметсте. Никакой реакции. С пустыми DE RE мах работать будет? Нету чем посмотреть что происходин да диференциальном выходи с мах... Тема новая. Два дня пытаюсь разобраться где накосячил. В железе или в софте. Пни в нужную сторону.

skipper4208, 2025-08-24:
Будь здоров.

closed 2025-08-24 by srg74